FRUK-Simulator/Simulator #4: skeleton 3d rendering code.

JamesWP, 2020-05-20:
create the classes responsible for the graphical rendering of the simulation environment

ginonott, 2020-05-20:
It might make sense to use https://github.com/react-spring/react-three-fiber. It leverages the declarative nature of JSX. A bonus is if the website is already using React, it could keep the code base a little more consistent and thus easier for people to get familiar with it.  A company that develops 3D casino games gave a talk about using React + Redux to drive their UI of the games as they found the one way data flow useful.

They did say that it wasn't performant for the games themselves however they were realistic casino card games which is most likely much more than we'll do.

Disclaimer: I've never used it and it may not be a good fit.

JamesWP, 2020-05-24:
This sounds very sensible, let’s figure out what sort of state management we would need e.g. redux. And what kind of events we would need

closed 2020-06-04 by JamesWP