FRUK-Simulator/Simulator #6: create react project template

JamesWP, 2020-05-20:


ginonott, 2020-05-20:
https://github.com/facebook/create-react-app should be able to fill this need. It hides all of the complexities of webpack, well maintained, and well documented. I also think we should go with https://github.com/facebook/create-react-app/tree/master/packages/cra-template-typescript which is the official typescript version. 

I've had great experience with using TS with React. It has some initial upfront learning curve for those only familiar with JS but the extra tooling that comes with it and the safety of types is worth it.

JamesWP, 2020-05-24:
Definitely agree, typescript is going to make us more productive later in the game

ginonott, 2020-05-28:
See #10

closed 2020-06-04 by ginonott