FRUK-Simulator/Simulator #22: Integrate simple fps indicator

JamesWP, 2020-06-04:


xycaleth, 2020-06-05:
I had a quick go at this but the FPS indicator ends up in the top-left of the page instead of being relative to the canvas. I'm not sure if that's a problem so I haven't submitted a PR but here's the change if anybody wants to continue with this further.
```diff
diff --git a/src/RobotSimulator/Sim3D.ts b/src/RobotSimulator/Sim3D.ts
index d5477a2..cd98ba6 100644
--- a/src/RobotSimulator/Sim3D.ts
+++ b/src/RobotSimulator/Sim3D.ts
@@ -1,11 +1,13 @@
 import * as THREE from "three";
 import { OrbitControls } from "three/examples/jsm/controls/OrbitControls";
+import Stats from 'three/examples/jsm/libs/stats.module.js';
 
 // This class is the top level 3D class and manages drawing and rendering the scene (and simulating it).
 // All interaction with react redux is done through its parent RobotSimulator, this class is abstracted from that.
 export class Sim3D {
   private scene: THREE.Scene;
   private renderer: THREE.Renderer;
+  private stats: Stats;
 
   private camera: THREE.PerspectiveCamera;
   private cameraControls: OrbitControls;
@@ -52,6 +54,10 @@ export class Sim3D {
     var helper = new THREE.GridHelper(1, 10);
     helper.scale.addScalar(10);
     scene.add(helper);
+
+    // Stats
+    var stats = (this.stats = Stats());
+    renderer.domElement.parentElement?.appendChild(stats.dom);
   }
 
   onresize() {
@@ -62,6 +68,7 @@ export class Sim3D {
 
   render() {
     this.cameraControls.update();
+    this.stats.update();
     this.renderer.render(this.scene, this.camera);
   }
```

JamesWP, 2020-06-08:
I think its still a good idea to have this, will be very useful for when we have more complex things going on to keep an eye on performance.
We've split the repo up a bit since this. Would love to see this in a PR.

JamesWP, 2020-06-08:
We can always hide this behind a developer mode later on in the project

closed 2020-06-12 by JamesWP