FRUK-Simulator/Simulator #24: When the application tab is not visible it should pause computation

JamesWP, 2020-06-07:
This encompasses the rendering and also the javascript vm.

sykes-j, 2020-06-07:
This optimisation is supposedly already provided by most browsers according to the [Page Visibility API](https://developer.mozilla.org/en-US/docs/Web/API/Page_Visibility_API) (see "Policies in place to aid background page performance"):

> Most browsers stop sending requestAnimationFrame() callbacks to background tabs or hidden <iframe>s in order to improve performance and battery life.

This `window.requestAnimationFrame()` API is used by our `Sim3D` class to drive the rendering and animation of the simulator. I assume the callback to this API is where we will call `vm.render()` or whatever, in which case both our VM and animation will be paused automatically.

However after testing the behaviour of this by adding a constant rotation to the `Sim3D.render()` function (code below to reproduce), I can see that the desired pausing only occurs when the simulator tab is not selected in the browser window. That is, the application is only paused when a different tab in the *same* browser window is switched to. If the user moves to a different application/browser window, then the application is not paused (presumably because the `visibilitychange` event is only fired by the browser on tab change, not window change).

```javascript
class Sim3D {
    private angle = 0;
    private radius = 1;
...
  render() {
    // Rotate the camera
    this.camera.position.x = this.radius * Math.cos(this.angle);
    this.camera.position.z = this.radius * Math.sin(this.angle);
    this.angle += 0.01;
    
    this.cameraControls.update();
    this.renderer.render(this.scene, this.camera);
  }
...
}
```

This behaviour is desirable in some ways: the simulator continues to play when opened side-by-side with another window (e.g. if the user has a chat application open on the other half of their screen). However it also means the user's device will be rendering in the background if they have the simulator application completely hidden by another window, which could slow their device down.

Potential solutions:
1. Investigate if there is a reliable way to detect when the browser window is hidden
    - [this stackoverflow question](https://stackoverflow.com/a/9502074) suggests `onblur` and `onfocus` can be used but is unreliable. Will we lose the ability to open the simulator side-by-side with another window by using this?
2. Implement a Play/Pause button in the `ControlPanel` so that users can choose to stop the simulation if their device is slowing down. I think this is more flexible than the solution proposed in (1).

Keen to hear any suggestions/solutions on this :)

JamesWP, 2020-06-07:
Nice, yes I think youre right about the `requestAnimationFrame` stuff. I think that will be sufficient for the graphics side of things. 
I think the only thing remaining is to ensure we have some auto pause stuff hooked up to the javascript vm when we get to that.. looks like @ginonott  has started here: #23

closed 2022-08-18 by JamesWP