FRUK-Simulator/Simulator #53: Add simple test back in

JamesWP, 2020-06-19:
There is a test which is commented out, we should figure out why it fails and how we can make it pass.

ginonott, 2020-06-20:
The reason why its failing is because SimulatorCore was being mocked out globally. Ideally we just want to mock out the Sim3D functions that render stuff. The problem is that if we invoke any file in the SimulatorCore package we get this:

```
    /home/ginonott/simulator/node_modules/three/examples/jsm/controls/OrbitControls.js:10
    import {
    ^^^^^^
```

The issue is that Jest runs through nodejs which cannot interop between ecma modules and CommonJS modules without special setup. 

There are a few options:
1. In SimulatorCore, switch from using the jsm version to the js version. However, when I tried that I lost all type definitions.
2. Try and get jest to transform node_modules into cjs modules
3. ???

JamesWP, 2020-06-20:
For some of the examples there are separate npm  modules. That might some this mixed mode import problem?

zhiquanyeo, 2020-06-20:
I think those separate npm modules (for threejs examples) are actually deprecated. I know that's true at least for the gltf loader. Installing the separate package for that gives a warning message that basically says "use the stuff in examples/jsm...."

closed 2020-09-18 by ginonott