FRUK-Simulator/Simulator #77: create blockly interface to query onscreen controller interface

JamesWP, 2020-07-07:
there should be blocks to query the current state of the controller.

currently, when clicking buttons on the onscreen controller interface motor control events are sent directly to the Sim3D robot handle. alongside this mode of operation there should be another mode where the controller's state is instead visible from the VM. there would also be blocks to access the state of each button on the controller.

JamesWP, 2020-07-07:
we should try to understand the existing First Tech challenge blocks which do this.

Casper-Oakley, 2020-07-08:
FTC block for reference (XML can be found in ftc blockly implementation):

[image]

closed 2020-07-27 by Casper-Oakley