FRUK-Simulator/Simulator #151: Power of motors and friction of wheels on the surface is not suitable (HIGH priority)

gavcradd, 2021-06-22:
Power % of motors need to be reduced. Only found it useable when set to 5% or 10%, 100% is ridiculous! Perhaps multiply current settings by 0.2 so that 100% becomes the old 20%? 

Also the robot at the moment behaves like it is on ice, removing power from the wheels does not significantly slow it down immediately.

MatthiasKreileder, 2021-06-22:
To make the robot move in a more realistic/user-friendly way we should probably aim for something like the following:

* 5%  motor power => crawling
* 100% motor power => 0.5m/second
* When the power gets sets back to zero => the robot should continue on its trajectory and then come to a stop after  around 0.1s and 1s

JamesWP, 2021-07-05:
I still think that there needs to be more friction applied to the wheels when the motor is set to 0 power. 

If you spin the robots with opposing motor powers and then set the motors to 0 power, it should take only a few moments to stop spinning.

JamesWP, 2021-07-05:
I had another attempt on top of what @MatthiasKreileder has done to continue to improve things.
https://github.com/FRUK-Simulator/SimulatorCore/pull/150/files

closed 2021-08-03 by JamesWP