FRUK-Simulator/Simulator #250: Completion time should be stored in order to display best time

anap1003, 2022-09-29:


ido-ran, 2022-10-03:
I think as a first step we need to move the timer measurement to redux store, currently it is stored inside the timer component, which is not accessible from the outside, right?

JamesWP, 2022-10-03:
There is an issue which might occur if we move the existing timer state directly into redux. The issue is that updating state in redux has a large overhead, for something frequent like timer updates (every 1/10 second) we might find that redux is a no go.

obviously we do need the timer information in redux for displaying in other areas on the ui, but not the whole timer state, perhaps just the last stopped time?

ido-ran, 2022-10-03:
I didn't mean update the actual count every 1/10 of a second, we just need to store enough information to be able to calculate the duration while running and have the final duration at the end.

This can be done by storing the start time and end time.
In order to support pause and continue we can also store the duration measured until the last pause.

The view can have a timer to constantly update the component state or even use `requestAnimationFrame` to do it as fast as possible.

The important part is to store enough information in redux to know the measurement.

JamesWP, 2022-10-03:
That sounds reasonable