FRUK-Simulator/Simulator #268: Thread to document different options of styling

cmanolescu, 2024-02-08:
Goal for this task is to document all the options on the table so the team can read through and comment on the options. 

A decision should be made on the style moving forward

yono10, 2024-02-13:
Thanks for the great summary @owen26 - amazing work as usual.

For me, any of these options beyond the vanilla CSS will be a learning curve 🙂 , but am up to use a more modern option anyway if the de facto approach used by web developers these days is to utilise one of those frameworks. And on that, with my minimal experience but just getting a feel from this summary, I would say my preference leans towards the SC (option 2), as it feels easier to understand / less magical ?

On the other hand - on CSS with modules, how would that look like? Would that be the route you'd recommend anyway if we were to choose option 1?

immutable-pro, 2024-02-13:
Amazing summary @owen26. I don't have a strong preference:

1. CSS modules before (mentioned in the cons of *Vanilla CSS*) ← I have extensively used it before.
2. Styled components ← I have extensively used it before.
3. Tailwind (I have a few reasons for disliking Tailwind, but I understand that it helps to move fast once you know it). ← We use it mixed with CSS modules in my current project, mostly because it has nice layout shortcuts.

Disclaimer for 1: I would never go with Vanilla CSS, but with CSS modules (the scoping reason alone, it is enough to consider).

I would not mind giving a go to **stylex** if any of us has experience with it. Otherwise, it is scary to adopt a Beta tech.

owen26, 2024-02-13:
Have discussed with the team and agreed to go with **Styled Component** as the choice of CSS framework for implementing this kobot new design.

closed 2024-02-13 by owen26